jostlowe/Pico-DMX #1: DMX Input / RDM possible?

hippyau, 2021-03-28:
Hi Jostein,  this is very cool!   What are the possibilities of receiving DMX, and possibly bi-directional RDM?

jostlowe, 2021-03-28:
Hi, Hippy!

As of now, the library does not support DMX reception or RDM. 

From what i've read in the documentation, the idea of using the PIO modules and DMA _should_ work quite well for DMX reception and RDM. I'm planning on implementing DMX reception for this library, but it will probably happen sometime after the RP2040 gets official Arduino core support. 

RDM is quite a complex piece of protocol magic, and I'm not sure if it will be implemented anytime soon.

GiacomoCappuccini, 2021-05-24:
Hi Jostein, any news regarding receiving DMX?  I'm a beginner and I was looking for examples for receiving DMX using rpi pico.

jostlowe, 2021-05-25:
Hi, Giacomo, and welcome :)

The official Arduino support for the RaspberryPi Pico was just released, so I'm currently working on porting the library over to the Arduino platform. As soon as i'm done porting the library, I'll have a look at adding support for a DMX receiver.

Keep an eye out in the coming month!

jostlowe, 2021-05-30:
DMX input was just added in version 2.0.0 of the library! Check it out :)

Please open an issue if you find any bugs in the code. Happy hacking, @GiacomoCappuccini and @hippyau

closed 2021-05-30 by jostlowe

hippyau, 2021-06-04:
That is great!  Thanks :)