jostlowe/Pico-DMX #6: Timing parameters

sstaub, 2021-06-17:
How does this library handle timings like Mark and Break?

jostlowe, 2021-06-17:
Hi! Could you elaborate a bit further on your question? Are you referring to how the library handles Mark and Break conditions when receiving or when sending?

sstaub, 2021-06-17:
Yes, I can't see anything in the code how you handle the Break and MAB (Mark after Break).

jostlowe, 2021-06-21:
The handling of MAB and break conditions is not explicitly handled in the C code itself. The Programmable IO (PIO) modules handle the specific timing requirements for the DMX protocol. The assembly code for DMX output and input can be found [here](extras/). 

For DMX output, the PIO module automatically transmits a Break and MAB every time a new frame is transmitted. As soon as the Break and MAB are transmitted, the PIO module starts pulling bytes from memory using DMA.

For DMX input, The PIO actively checks for a break condition on the DMX line. If the line is continuously kept low for more than 90 microseconds, a break condition is asserted. The PIO then waits for the MAB and first start bit before it starts reading bytes from the frame.

sstaub, 2021-06-21:
Ok, this helps. Is there a way to made Break and MAB timing configurable easily?

jostlowe, 2021-06-21:
Unfortunately, no...

The timing parameters are embedded in the PIO assembly code, so configuration of Break and MAB timing requires you to edit and re-assemble the PIO assembly code. 

It is possible, but not particularly easy

sstaub, 2021-06-21:
Thank you. I will test the library soon.

closed 2021-06-21 by sstaub