jostlowe/Pico-DMX #7: Adding IRQ to indicate a full frame has been received

larry104, 2021-07-04:
It would be great feature addition to alert the CPU with an interrupt flag when a complete frame has been received without any errors.

What do you think?

functionpointer, 2021-07-12:
You might be interested in https://github.com/jostlowe/Pico-DMX/pull/8

larry104, 2021-07-12:
that's pretty cool ... Now the 'only" thing missing is a heart beat to synchronize for instance a strobe feature between fixtures. I guess the DMA_IRQ0 can't be used for that as different fixtures might have different addresses and so the IRQ will come at different times or do I see that wrong?

functionpointer, 2021-07-12:
You are correct, it will come at slightly different times. Unfortunately, DMX doesn't really have an "end-of-packet" that can be reliably detected. This is because there is no dedicated end signal and MTBF and MTBP are both allowed to be extremely long.

For this reason, we can only rely on detecting the start of the DMX packet and then count until our channels of interest come around.

For synchronization you could have multiple fixtures trigger their strobes from the same channel. Alternatively you can probably get the synchronization within a couple milliseconds just by using a DMX master with a short MTBF and by having the strobe channels clustered together.

functionpointer, 2021-07-12:
Also i just thought of something: Just because you read up to channel 52 doesn't mean you can't just use channels 1-3!

You can configure all your fixtures to read to the same end channel. That way the IRQ will trigger at the same time on all of them. 

This method does waste some memory, but the current pull request is super wasteful anyway. Thankfully the pico has lots of memory.

larry104, 2021-07-13:
Yeah, that is a good idea. I will definitely give this a try. My current Microchip Pic implementations sucks big time because a driving an oled display, dmx and fan control is just too much. The Pico is a paradise with the 2 cores and the PIO's ;-)

jostlowe, 2021-07-13:
@functionpointer your reasoning seems correct! Triggering an IRQ when a full frame has been received is not that easy, as there really is no way of knowing when a frame ends apart from waiting for the Break Condition of the next frame. While staying within the DMX-spec, that could lead to some serious latency between the controller and the fixtures in certain cases. 

For a pixelated LED square ceiling (the one in my GitHub pfp, actually) we needed to synchronize the strobe feature for all 104 squares. The dude who designed the DMX-receivers made all the squares listen to the same 2 channels for strobe control, kind of like @functionpointer suggests. It worked really well for us :) 

@larry104 let us know how the strategy worked for you!

closed 2021-08-21 by jostlowe

kripton, 2021-12-08:
See also #14