jostlowe/Pico-DMX #25: DMX for PICO without PIO

Gavin-Perry, 2022-03-16:
RP2040 Pico using Arduino 2.0 IDE on Win11
I like the way your library makes DMX easy!
I've used all the PIOs for my 16 PWM LED lights. Is there a way for this lib to work with UART instead of PIO?
Also I'm looking for channel 400 - 512 for this particular app. Has issue re Start Chanel been fixed yet?

Gavin-Perry, 2022-03-16:
Sorry, I'm mixing up PWM slices (8) with the PIO blocks.
I suppose I can use all the PWM slices AND a PIO for the DMX
So only remaining issue is getting to channel 512 and from comments on that issue it looks like it might be fixed.

kripton, 2022-03-17:
You are right, the PWM slices and the PIOs are two completely independent pieces of hardware. They do share the internal bus of course, but with "just" some DMX and PWM it will be far from saturated.
The start_channel-issue is still relevant if we are talking about DMX input: https://github.com/jostlowe/Pico-DMX/issues/15. You can receive and use all channels of a universe. However, you need to receive the complete universe and find the relevant channel yourself (= you allocate and provide a 513 byte large buffer and use the bytes relevant for you). The start_channel parameter was meant so if you're only interested in channels 500 t0 512, you would only need to provide a 12 byte large buffer and save some memory.

Gavin-Perry, 2022-03-17:
Thanks for your reply.
I set DMXinput up to receive all channels. That's the easiest and there is
plenty of memory in a pico.
I transfer to 16 PWM channels in the selected group from 400-415 (group 0)
up to 496-512 (group 7)
That way I only needed 3 address bits (which is all I had left on my
RP2040-Zero anyway).
Nice that a 1" square can control 16 lights without breaking a sweat. Loop
code is just

for(i=0; i<8; i++) {
 pwm_set_counter
<https://raspberrypi.github.io/pico-sdk-doxygen/group__hardware__pwm.html#ga5179ec971146ab611b9232587e945769>
(i,
DMXbuffer[START_CHAN+i);
}
delay(UPDATE_RATE); // Could look for next buffer full but who cares

For testing purposes I set up DMXoutput in loop1(). Why not run that other
processor?

Thank you for the great library!

Gavin-Perry, 2022-03-18:
> <jostlowe/Pico-DMX/issues/25/1070518560

Mis remembered. Actually I used
  for (int i = 0; i <NUM_CHANNELS; i++) {  // For all 16 channels
    pwm_set_gpio_level(i, lookupGamma[buffer[StartChannel+i]] );
 // Or is buffer idexeded to channel #??
  }
}
to set the PWM counter for each GPIO pin

closed 2022-03-18 by Gavin-Perry

Gavin-Perry, 2022-03-24:
I'm not getting channel 1 to behave properly when I use a real DMXoutput
device (Chauvet)
Channel alignment works when I loop back DMXOutput to DMXInput (using the
other processor, very handy.

My code attached.. Trying to get channel 1 to work right and not offset all
the channels by one.