jostlowe/Pico-DMX #26: getting problem in running inbuilt sample code RGB_input_async in pico in arduino....

sagarpateldt, 2022-04-19:
I receive this on comport while run inbuilt sample code RGB_input_async in pico in arduino....
code description is this..
* Description: 
 * Starts a DMX Input on GPIO pin 0 and read channel 1-3 repeatedly
 */
but it shows four lines as show below and in first column 255 value randomly occurred. What is the problem?

09:08:04.175 -> Received packet: 0, 0, 0, 0, 
09:08:04.211 -> Received packet: 0, 0, 0, 0, 
09:08:04.257 -> Received packet: 0, 0, 0, 0, 
09:08:04.304 -> Received packet: 0, 0, 0, 0, 
09:08:04.351 -> Received packet: 0, 0, 0, 0, 
09:08:04.351 -> Received packet: 0, 0, 0, 0, 
09:08:04.398 -> Received packet: 0, 0, 0, 0, 
09:08:04.445 -> Received packet: 0, 0, 0, 0, 
09:08:04.492 -> Received packet: 0, 0, 0, 0, 
09:08:04.539 -> Received packet: 0, 0, 0, 0, 
09:08:04.586 -> Received packet: 0, 0, 0, 0, 
09:08:04.633 -> Received packet: 0, 0, 0, 0, 
09:08:04.680 -> Received packet: 0, 0, 0, 0, 
09:08:04.680 -> Received packet: 0, 0, 0, 0, 
09:08:04.725 -> Received packet: 0, 0, 0, 0, 
09:08:04.772 -> Received packet: 0, 0, 0, 0, 
09:08:04.818 -> Received packet: 0, 0, 0, 0, 
09:08:04.865 -> Received packet: 0, 0, 0, 0, 
09:08:04.913 -> Received packet: 255, 0, 0, 0, 
09:08:04.957 -> Received packet: 0, 0, 0, 0, 
09:08:04.957 -> Received packet: 0, 0, 0, 0, 
09:08:05.004 -> Received packet: 0, 0, 0, 0, 
09:08:05.050 -> Received packet: 0, 0, 0, 0, 
09:08:05.097 -> Received packet: 0, 0, 0, 0, 
09:08:05.144 -> Received packet: 0, 0, 0, 0, 
09:08:05.192 -> Received packet: 0, 0, 0, 0, 
09:08:05.238 -> Received packet: 0, 0, 0, 0, 
09:08:05.285 -> Received packet: 0, 0, 0, 0, 
09:08:05.285 -> Received packet: 0, 0, 0, 0, 
09:08:05.332 -> Received packet: 0, 0, 0, 0, 
09:08:05.379 -> Received packet: 0, 0, 0, 0, 
09:08:05.426 -> Received packet: 0, 0, 0, 0, 
09:08:05.475 -> Received packet: 0, 0, 0, 0, 
09:08:05.521 -> Received packet: 0, 0, 0, 0, 
09:08:05.569 -> Received packet: 0, 0, 0, 0, 
09:08:05.569 -> Received packet: 0, 0, 0, 0, 
09:08:05.616 -> Received packet: 255, 0, 0, 0, 
09:08:05.660 -> Received packet: 0, 0, 0, 0, 
09:08:05.707 -> Received packet: 0, 0, 0, 0, 
09:08:05.754 -> Received packet: 0, 0, 0, 0, 
09:08:05.800 -> Received packet: 0, 0, 0, 0, 
09:08:05.848 -> Received packet: 0, 0, 0, 0, 
09:08:05.895 -> Received packet: 0, 0, 0, 0, 
09:08:05.895 -> Received packet: 0, 0, 0, 0, 
09:08:05.942 -> Received packet: 0, 0, 0, 0, 
09:08:05.989 -> Received packet: 0, 0, 0, 0, 
09:08:06.037 -> Received packet: 0, 0, 0, 0, 
09:08:06.083 -> Received packet: 0, 0, 0, 0, 
09:08:06.130 -> Received packet: 0, 0, 0, 0, 
09:08:06.176 -> Received packet: 0, 0, 0, 0, 
09:08:06.225 -> Received packet: 0, 0, 0, 0, 
09:08:06.225 -> Received packet: 0, 0, 0, 0, 
09:08:06.271 -> Received packet: 0, 0, 0, 0, 
09:08:06.319 -> Received packet: 0, 0, 0, 0, 
09:08:06.369 -> Received packet: 0, 0, 0, 0, 
09:08:06.412 -> Received packet: 0, 0, 0, 0, 
09:08:06.459 -> Received packet: 0, 0, 0, 0, 
09:08:06.507 -> Received packet: 0, 0, 0, 0, 
09:08:06.507 -> Received packet: 0, 0, 0, 0, 
09:08:06.551 -> Received packet: 0, 0, 0, 0, 
09:08:06.596 -> Received packet: 0, 0, 0, 0, 
09:08:06.643 -> Received packet: 0, 0, 0, 0, 
09:08:06.691 -> Received packet: 0, 0, 0, 0, 
09:08:06.737 -> Received packet: 0, 0, 0, 0, 
09:08:06.784 -> Received packet: 0, 0, 0, 0, 
09:08:06.831 -> Received packet: 0, 0, 0, 0, 
09:08:06.831 -> Received packet: 0, 0, 0, 0, 
09:08:06.878 -> Received packet: 0, 0, 0, 0, 
09:08:06.925 -> Received packet: 0, 0, 0, 0, 
09:08:06.972 -> Received packet: 0, 0, 0, 0, 
09:08:07.018 -> Received packet: 0, 0, 0, 0, 
09:08:07.067 -> Received packet: 0, 0, 0, 0, 
09:08:07.114 -> Received packet: 0, 0, 0, 0, 
09:08:07.114 -> Received packet: 0, 0, 0, 0, 
09:08:07.160 -> Received packet: 0, 0, 0, 0, 
09:08:07.207 -> Received packet: 0, 0, 0, 0, 
09:08:07.253 -> Received packet: 0, 0, 0, 0, 
09:08:07.301 -> Received packet: 0, 0, 0, 0, 
09:08:07.349 -> Received packet: 0, 0, 0, 0, 
09:08:07.396 -> Received packet: 0, 0, 0, 0, 
09:08:07.396 -> Received packet: 0, 0, 0, 0, 
09:08:07.442 -> Received packet: 0, 0, 0, 0, 
09:08:07.489 -> Received packet: 0, 0, 0, 0, 
09:08:07.536 -> Received packet: 0, 0, 0, 0, 
09:08:07.581 -> Received packet: 0, 0, 0, 0, 
09:08:07.628 -> Received packet: 0, 0, 0, 0, 
09:08:07.670 -> Received packet: 0, 0, 0, 0, 
09:08:07.708 -> Received packet: 0, 0, 0, 0, 
09:08:07.755 -> Received packet: 0, 0, 0, 0, 
09:08:07.802 -> Received packet: 0, 0, 0, 0, 
09:08:07.802 -> Received packet: 0, 0, 0, 0, 
09:08:07.850 -> Received packet: 0, 0, 0, 0, 
09:08:07.897 -> Received packet: 0, 0, 0, 0, 
09:08:07.944 -> Received packet: 0, 0, 0, 0, 
09:08:07.992 -> Received packet: 0, 0, 0, 0, 
09:08:08.036 -> Received packet: 0, 0, 0, 0, 
09:08:08.082 -> Received packet: 0, 0, 0, 0, 
09:08:08.129 -> Received packet: 0, 0, 0, 0, 
09:08:08.129 -> Received packet: 0, 0, 0, 0, 
09:08:08.175 -> Received packet: 0, 0, 0, 0, 
09:08:08.221 -> Received packet: 0, 0, 0, 0, 
09:08:08.268 -> Received packet: 0, 0, 0, 0, 
09:08:08.315 -> Received packet: 0, 0, 0, 0, 
09:08:08.363 -> Received packet: 0, 0, 0, 0, 
09:08:08.411 -> Received packet: 255, 0, 0, 0, 
09:08:08.459 -> Received packet: 0, 0, 0, 0, 
09:08:08.459 -> Received packet: 0, 0, 0, 0, 
09:08:08.506 -> Received packet: 0, 0, 0, 0, 
09:08:08.553 -> Received packet: 0, 0, 0, 0, 
09:08:08.601 -> Received packet: 0, 0, 0, 0, 
09:08:08.649 -> Received packet: 0, 0, 0, 0, 
09:08:08.695 -> Received packet: 0, 0, 0, 0, 
09:08:08.695 -> Received packet: 0, 0, 0, 0, 
09:08:08.743 -> Received packet: 0, 0, 0, 0, 
09:08:08.791 -> Received packet: 0, 0, 0, 0, 
09:08:08.837 -> Received packet: 0, 0, 0, 0, 
09:08:08.884 -> Received packet: 0, 0, 0, 0, 
09:08:08.930 -> Received packet: 0, 0, 0, 0, 
09:08:08.978 -> Received packet: 0, 0, 0, 0, 
09:08:09.025 -> Received packet: 0, 0, 0, 0, 
09:08:09.025 -> Received packet: 0, 0, 0, 0, 
09:08:09.073 -> Received packet: 0, 0, 0, 0, 
09:08:09.120 -> Received packet: 0, 0, 0, 0, 
09:08:09.167 -> Received packet: 0, 0, 0, 0, 
09:08:09.214 -> Received packet: 0, 0, 0, 0, 
09:08:09.263 -> Received packet: 0, 0, 0, 0, 
09:08:09.308 -> Received packet: 0, 0, 0, 0, 
09:08:09.308 -> Received packet: 0, 0, 0, 0, 
09:08:09.354 -> Received packet: 0, 0, 0, 0, 
09:08:09.401 -> Received packet: 0, 0, 0, 0, 
09:08:09.448 -> Received packet: 0, 0, 0, 0, 
09:08:09.497 -> Received packet: 0, 0, 0, 0, 
09:08:09.541 -> Received packet: 0, 0, 0, 0, 
09:08:09.591 -> Received packet: 0, 0, 0, 0, 
09:08:09.591 -> Received packet: 0, 0, 0, 0, 
09:08:09.635 -> Received packet: 0, 0, 0, 0, 
09:08:09.681 -> Received packet: 0, 0, 0, 0, 
09:08:09.729 -> Received packet: 0, 0, 0, 0, 
09:08:09.777 -> Received packet: 0, 0, 0, 0, 
09:08:09.824 -> Received packet: 0, 0, 0, 0, 
09:08:09.875 -> Received packet: 0, 0, 0, 0, 
09:08:09.875 -> Received packet: 0, 0, 0, 0, 
09:08:09.920 -> Received packet: 0, 0, 0, 0, 
09:08:09.964 -> Received packet: 0, 0, 0, 0, 
09:08:10.011 -> Received packet: 0, 0, 0, 0, 
09:08:10.059 -> Received packet: 0, 0, 0, 0, 
09:08:10.106 -> Received packet: 0, 0, 0, 0, 
09:08:10.153 -> Received packet: 0, 0, 0, 0, 
09:08:10.201 -> Received packet: 0, 0, 0, 0, 
09:08:10.201 -> Received packet: 0, 0, 0, 0, 
09:08:10.248 -> Received packet: 0, 0, 0, 0, 
09:08:10.294 -> Received packet: 0, 0, 0, 0, 
09:08:10.341 -> Received packet: 0, 0, 0, 0, 
09:08:10.387 -> Received packet: 0, 0, 0, 0, 
09:08:10.434 -> Received packet: 0, 0, 0, 0, 
09:08:10.482 -> Received packet: 0, 0, 0, 0, 
09:08:10.482 -> Received packet: 255, 0, 0, 0, 
09:08:10.530 -> Received packet: 0, 0, 0, 0, 
09:08:10.577 -> Received packet: 0, 0, 0, 0, 
09:08:10.623 -> Received packet: 0, 0, 0, 0, 
09:08:10.670 -> Received packet: 0, 0, 0, 0, 
09:08:10.717 -> Received packet: 0, 0, 0, 0, 
09:08:10.765 -> Received packet: 0, 0, 0, 0, 
09:08:10.765 -> Received packet: 0, 0, 0, 0, 
09:08:10.813 -> Received packet: 0, 0, 0, 0, 
09:08:10.861 -> Received packet: 0, 0, 0, 0, 
09:08:10.909 -> Received packet: 0, 0, 0, 0, 
09:08:10.956 -> Received packet: 0, 0, 0, 0, 
09:08:11.004 -> Received packet: 0, 0, 0, 0, 
09:08:11.050 -> Received packet: 0, 0, 0, 0, 
09:08:11.097 -> Received packet: 0, 0, 0, 0, 
09:08:11.097 -> Received packet: 0, 0, 0, 0, 
09:08:11.142 -> Received packet: 0, 0, 0, 0, 
09:08:11.189 -> Received packet: 0, 0, 0, 0, 
09:08:11.237 -> Received packet: 0, 0, 0, 0, 
09:08:11.284 -> Received packet: 0, 0, 0, 0, 
09:08:11.332 -> Received packet: 0, 0, 0, 0, 
09:08:11.379 -> Received packet: 0, 0, 0, 0, 
09:08:11.379 -> Received packet: 0, 0, 0, 0, 
09:08:11.427 -> Received packet: 0, 0, 0, 0, 
09:08:11.475 -> Received packet: 0, 0, 0, 0, 
09:08:11.523 -> Received packet: 255, 0, 0, 0, 
09:08:11.569 -> Received packet: 0, 0, 0, 0, 
09:08:11.617 -> Received packet: 0, 0, 0, 0, 
09:08:11.617 -> Received packet: 0, 0, 0, 0, 
09:08:11.665 -> Received packet: 0, 0, 0, 0, 
09:08:11.713 -> Received packet: 0, 0, 0, 0,

flavioamieiro, 2023-05-19:
I have the exact same issue with the `RGB_input_async` example. 

If I have the DMX controller sending data, the buffer is always all `0`s. If I stop sending DMX data, the code correctly detects that we're not receiving DMX packets. I also tried using the optional callback function as an argument to `read_async` but, again, the buffer is always empty.

If I run the synchronous `read` function, `buffer` is always populated as expected, and I can read the correct values for all channels.

@sagarpateldt did you find a solution for this? Maybe @jostlowe can also give me a pointer to what I could be doing wrong here?

Thanks a lot for the library, by the way!

jostlowe, 2023-05-21:
@sagarpateldt could you share your solution if you found one?

flavioamieiro, 2023-06-02:
I'm sorry it took me a while to come back to this. I was looking at the issue again today and realized that the problem is that `GPIO0` (the default input pin in the example) is also `UART_TX`. I confirmed this was the issue by not sending data through (or initializing) the UART interface and it worked.

<details>
<summary>example without UART</summary>

```
#include <Arduino.h>
#include "DmxInput.h"
DmxInput dmxInput;

#define START_CHANNEL 1
#define NUM_CHANNELS 3

volatile uint8_t buffer[DMXINPUT_BUFFER_SIZE(START_CHANNEL, NUM_CHANNELS)];

void setup()
{
    // Setup our DMX Input to read on GPIO 0, from channel 1 to 3
    dmxInput.begin(0, START_CHANNEL, NUM_CHANNELS);
    dmxInput.read_async(buffer);

    // Setup the onboard LED so that we can blink when we receives packets
    pinMode(LED_BUILTIN, OUTPUT);
}

void loop()
{
    delay(30);

    if(millis() > 100+dmxInput.latest_packet_timestamp()) {
        return;
    }
    for (uint i = 0; i < sizeof(buffer); i++)
    {
      if (buffer[i] > 127) {
        delay(10);
        digitalWrite(LED_BUILTIN, HIGH);
        delay(10);
        digitalWrite(LED_BUILTIN, LOW);
      }
    }
}

```

</details>

Changing the pin to `GPIO15` fixed the issue completely, even with UART enabled. 

The following patch changes all examples not to use `GPIO0` or `GPIO1`, as they are used for UART. Please feel free to merge it if you think it makes sense @jostlowe , but please take a closer look at the `parallel_output` example, as the initialization code is not really nice the way I patched it.

Again, thanks a lot for the great library :)

```patch
From 77f2a2102ae6d44803bed29da578b58c49098080 Mon Sep 17 00:00:00 2001
From: Flavio Amieiro <amieiro.flavio@gmail.com>
Date: Thu, 1 Jun 2023 22:17:05 -0300
Subject: [PATCH] Changes DMX pin on examples so they don't conflict with UART

While investigating #26 I realized that the default pin in the examples
(pin 0) is also the TX pin for UART in the microcontroller. Since the
`RGB_async_input` example sends the received data through UART, this was
causing issues reading the data. This commit changes the code so users
who run the async example don't run into the same issue, but also
changes the other examples so sending and receiving data through UART
won't affect the DMX input/output.
---
 examples/RGB_input/RGB_input.ino             | 10 +++++-----
 examples/RGB_input_async/RGB_input_async.ino | 10 +++++-----
 examples/highlight_all/highlight_all.ino     | 10 +++++-----
[... 106 lines cut ...]
 void setup()
 {
-    // Start the DMX outputs on GPIO-pins 0-7.
+    // Start the DMX outputs on GPIO-pins 2-9.
     // Only 4 outputs can run on a single PIO instance, so
     // the 8 outputs are divided onto the two PIO instances
     // pio0 and pio1
 
-    for (int i = 0; i < 4; i++)
+    for (int i = 2; i < 6; i++)
     {
-        dmxOutputs[i].begin(i, pio0);
+        dmxOutputs[i-2].begin(i, pio0);
     }
-    for (int i = 4; i < 8; i++)
+    for (int i = 6; i < 10; i++)
     {
-        dmxOutputs[i].begin(i, pio1);
+        dmxOutputs[i-2].begin(i, pio1);
     }
 }
 
-- 
2.39.2
```