jostlowe/Pico-DMX #27: Support for differential outputs (no RS-485 module necessary)

GlenHertz, 2022-04-19:
Fantastic project!  Is it possible to add support for sending differential outputs in order to not have to purchase an RS-485 module?  As far as I can tell using 2 pins to generate the differential signals (one the negation of the other) should work for DMX signals and the PIOs should be able to send them simultaneously.  The RP2040 is using 3.3V logic and for the RS-485 receivers I've looked at they take ±200 mV for an ON/OFF bit so it should work.  I'd like to use something like the RP2040 XIAO for a tiny DMX controller (send only).

kripton, 2022-06-14:
Oh, what would probably also work:
Have TWO PIO state machines per DMX universe, one generates the "positive signal", the other one the inverse. If both operate on the same data, have the same clock and are started simultaneously (not by the CPU as instructions but maybe via an interrupt?) that might work as well

6by9, 2022-06-14:
An RS485 transmitter is meant to drive a minimum 1.5V differential signal into a 54ohm load. Whilst the Pico will put out a 3.3V differential signal, the drive current may be limited (max 12mA according to the datasheet, and max 50mA for all GPIOs).

Yes an RS485 receiver is required to be sensitive enough for a 200mV differential signal, but launching a low signal onto a long line is possibly not the best plan. It would offer a cheap solution though.

6by9, 2022-06-16:
One of my hardware colleagues here at Pi Towers is also involved in lighting for various theatres in the area, so I've had a quick chat with him.

The spec 12mA is fairly conservative, and that is for the output voltage to reach the specified VOH and VOL voltages on the pins.

If driving the two sides of the line from GPIOs, then you're looking at 3.3V being driven down the line, through the 100ohm terminator, and coming back again. 3.3 / 100 = 33mA.

Put a 47ohm resistor in series with each leg before putting it on the wire, you halve the voltage down the line, but also halve the current to 16..8mA. It also gives some protection the output against shorts, and acts as a terminator for any reflections coming back up the line.
If driving a 54ohm load then you'd be looking at a 108ohm load total, which would want 30.5mA. You may not achieve the full 1.5V of the spec, but it should be pretty reasonable.

The current per universe is constant as whenever one GPIO is sourcing, the other is sinking. If running at 16.8mA, then 4 universes be plausible at 67mA total. Running 8 universes without buffering may be pushing it.

Looking at your rp2040-dmxsun project, an output board that takes 4 GPIOs and produces 2 universes without buffers would be simple for a reduced cost board. You obviously have very little ESD or other protection on those ports compared to an RS485 transceiver, but it would work in many situations.

kripton, 2022-06-22:
I can confirm that this is working. I implemented it the way I proposed in the dmxsun project (= generating the bit patterns "in software" = by the CPU) and having it simply clocked out via one PIO. = Not how Pico-DMX does it): https://github.com/kripton/rp2040-dongle/tree/DirectDifferentialOutput

I have only one Cameo ThunderWash 100 RGB connected at the moment and it actually already works when only GND and XLR pin 3 (positive) is connected. The fixture flickers from time to time but it's really rare. Connecting the XLR pin 2 (negative) makes it rock solid. no RS-485 driver used on my board.

Ideally, there would be a video here as proof but it's too messy around here right now, so a picture needs to be enough (you need to believe me that the fixture is connected to the XLR cable one can see in the picture):
[image]

So, yes, regarding the rp2040-dmxsun project, this would be a very cheap output-board. ESD and protections might not be toooo bad (Rs in series to the data lines, Z-diodes to clamp the voltage, ...). It might not meet the DMX512-standard but as the OP said, it might help in some situations and is really cheap.

Hopefully I can provide some oscilloscope-screenshots comparing what's on the line with the direct GPIO outs vs. one RS-485 driver.

I'm still thinking about this feature (differential output) could be implemented in Pico-DMX's current implementation but in my head, it's not easy. Yes, as I already mentioned, using two State-Machines per universe might work. I'd like to try it out at least ;)

drewTheNerd, 2022-06-25:
Just discovered this library (havent used it yet) but it looks super exciting!

As for the whole trying to avoid using an RS485 driver, I don't see why one would really need to do so. Max485 chips are incredibly cheap on Aliexpress and even on amazon aren't too bad. Using the right chip will help you avoid implementation issues down the line when lights are flickering because your Pi can't put out enough power.

That said, the fact that you figured it out without said chip is quite impressive none the less and definitely an option for the absolute easiest option.

GlenHertz, 2022-10-28:
@kripton amazing work!  I'll have to try it out.  In my application I'm just driving a DMX input on a DMX Controller and want to put it in a small surface mount RG45 box.  This will be ideal.  Thanks again!

kripton, 2022-10-28:
@GlenHertz : You're welcome, keep us posted on the progress and if it worked!
And I have to agree with @e120guru : DMX was not meant to be driven directly by IO pins. You have 0 protection and it might not work. Don't use it for DMX input, only for output.
I would say that the "0 protection" can be ignored since the Pico is reeealllyyy cheap :D If it breaks, just replace it. Just bad if it happens during a show ...

jostlowe, 2023-03-09:
I love the discussion, and while the possibility to drive a DMX line without a differential driver is alluring, I think it will probably cause more issues than benefits for the users. 

I'm closing the issue for now, but I'll leave the discussion for anyone out there wanting to experiment with "unsafe" DMX :)

closed 2023-03-09 by jostlowe