jostlowe/Pico-DMX #31: Question: dma support and power saving tipps?

NeoCortex97, 2022-07-11:
This is not about a problem with the library, but i wanted to dma to continually send my dmx universes and was wondering if it would be possible with this library.

As a short disclaimer: I did not look into anythin how dma works with the pico arduino libraries, so please don't hate me, if this is dumb, but I did not find a single mention of dma besides in the beginning of the README.

I want to build a dmx master that is working on battery power and that is mainly used for pretty static lighting, so i was hoping to put the RP2040 into a lower power state that still leaves dma and pio components running, but that saves at least a little energy.

If your answer is that dma is not supported and that there is no low power mode that suits my expectations (Still reading up on the rp2040, so i don't know yet), do you have ideas how I could save energy instead?
Not sending is not rally an option, since one of the fixtures randomly freaks out, if it does not receive a packet for some time. Also times until freakouts were not really predictable and vary form about 200ms to 3 hours

jostlowe, 2022-07-12:
Hi! Excellent question. No need to fear any hate here :) I'll do my best to elaborate, and maybe some of the other smart ducks here can correct me if something is wrong. 

The way this library works is that it uses some processor instructions to set up a PIO instance to transmit a single DMX frame and then uses a DMA channel to feed the DMX channels/bytes into the PIO. For each single frame transmitted, the processor needs to restart the PIO program in the PIO instance and also reconfigure the DMA channel. 

In practice, this means that when using this library, you need to run at least a few processor cycles every time you want to send DMX frame. On the other hand, it shouldn't be that hard to configure the RP2040 so that an IRQ is fired when the DMX frame is done transmitting. The processor can be put into a low power mode while the PIO and DMA pump out the DMX frame onto the GPIOs and be awakened by the IRQ when it is time to send a new frame. 

Check out `DMA_IRQ_0` and `DMA_IRQ_1` in the datasheet for the RP2040 

This should at least save you a lot of idling time on your processor

Let me know if this answered your question!

NeoCortex97, 2022-07-12:
This helps alot, thank you verry much. I think in the long run would be adding an example with the dma interrupt a good idea.

Now I have just another Question that I could look up myself, but you might have written down yourself.
Is there any hint in the spec that says how many frames per second should be sent? Or at least how long you can take between frames. I might add a little more intelligence to my master, so that it can play fades smoothly and starts sending frames slower if it is just displaying static frames.
Most of my fixtures just continue minding their own business if no frames are sent, so for most fixtures sake, i could just send the last frame and go to sleep indefinitely. But one Fixture (a cheap movinghead) start freaking out and doing random movements after a time, so i need to continually send frames. I did some experiments and measured about 200ms until it starts acting up and was interested, if the spec had something to say about that.

If you don't know the answer it's okay, I'll just read the spec myself.

jostlowe, 2022-07-14:
The most reasonable behavior for a fixture is to always enact the last frame recieved. It would be a bit unfortunate if you were doing a live concert or theatre, and when the DMX transmitter died, the entire stage went dark after 5 seconds of no data :)

From what i remember, the DMX spec does not specify a timeout. Most fixtures still have some kind of indicator to show when signal is lost, but how many packets have to be lost within what timeframe is simply picked by the manufacturers. Most modern DMX transmitters transmit frames at maximum rate even when there is no change to the frame.

For your case, I'd set up some experiments and see how long you can wait in between frames before your budget moving head starts freaking out :) 5 frames per seconds is definetly more battery saving than 44 frames per second which is the standard transmission rate

NeoCortex97, 2022-07-14:
Thank you very much for your help! I think that I can implement my project now, so i will mark this Issue as closed :-D

closed 2022-07-14 by NeoCortex97