jostlowe/Pico-DMX #34: Can you provide a cmake example including your library

karamandrew, 2022-09-12:
Hello, I have been struggling a while to include your library. New issues appeared everywhere using the sdk.

kripton, 2022-09-12:
Can you please give us some more info about the environment you plan to use the library in? IDE, SDK, Platform, Hardware?

karamandrew, 2022-09-12:
Yes sorry I posted my question in a bad mood with no real hopes of answers, thank you for your fast reply though. 

I am building it on Ubuntu and I am currently trying to use the library at it's minimal capabilities, which would be to turn all channels on thanks to a main file calling the library.
I want to build it using the SDK because I'll have to run multiple stuffs on my pico and I want to have the best control to optimize its performance.

The real problem is that I am totally new to cmake (I used visual studio before to build my c++ projects) and even with cmake tutorials, there is something I am missing...

I do include the .cmake file that is furnished with the library in my main CMakeLists but the building still don't find the DmxOutput.h.

Here is the error: 
**"
Scanning dependencies of target test_project
[ 18%] Building CXX object CMakeFiles/test_project.dir/main.cpp.obj
/home/pico/test/main.cpp:1:10: fatal error: DmxOutput.h: No such file or directory
    1 | #include <DmxOutput.h>
      |          ^~~~~~~~~~~~~"**

And here is my CMake file:
"
**cmake_minimum_required(VERSION 3.13)

include($ENV{PICO_SDK_PATH}/pico_sdk_init.cmake)

project(test_project C CXX ASM)

set(CMAKE_CXX_STANDARD 17)

pico_sdk_init()

include($ENV{HOME}/pico/libraries/Pico-DMX/interfaceLibForPicoSDK.cmake)

add_executable(test_project
    main.cpp
)

target_link_libraries(test_project pico_stdlib hardware_pio hardware_dma)

pico_add_extra_outputs(test_project)"**

unitware, 2022-11-05:
try `#include "DmxOutput.h"`

and add picodmx to your linked libraries
`target_link_libraries(test_project pico_stdlib hardware_pio hardware_dma picodmx)`

jostlowe, 2023-03-09:
Did the answer from @unitware solve the issue @karamandrew ?