jostlowe/Pico-DMX #35: Issue Using Pico-DMX together with Adafruit_NeoPXL8

HerbertHalamek, 2022-10-13:
When using "**Pico-DMX**" together with "**Adafruit_NeoPXL8**" on Raspberry Pi Pico with Arduino (earlephilhower core) the processor behaves unpredictable (normally blocks all communication).
The root source for this problem is that both libraries use the same Interrupt (IRQ0) for DMA:

**Pico-DMX / DmxInput.cpp / DmxInput::read_async**
       dma_channel_set_irq0_enabled(_dma_chan, true);
       irq_set_exclusive_handler(DMA_IRQ_0, dmxinput_dma_handler);
       irq_set_enabled(DMA_IRQ_0, true);

**Adafruit_NeoPXL8 / Adafruit_neoPXL8.cpp / Adafruit_NeoPixel::begin**
	irq_set_exclusive_handler(DMA_IRQ_0, dma_finish_irq);
	dma_channel_set_irq0_enabled(dma_channel, true);
	irq_set_enabled(DMA_IRQ_0, true);

Changing one of the two to IRQ1 in all three code lines solves the problem and everything runs like a charm.

I don't know whether it is possible to sort of claim an IRQ like for the DMA channels and the PIO state machines.
If not it would be very desirable to add a parameter <IRQ-index> to the read_async method. Which allows to select the IRQ by the application in case of conflicts.
Same is true for the DmxInput::read method. DmxOutput not tested yet.

I will also report same thing to Adafruit.

kripton, 2022-10-24:
I agree that it might be helpful to be able to change this using an additional parameter or `#define`. PRs welcome :)

kripton, 2022-10-24:
Duplicate of #33 ?

jostlowe, 2022-10-25:
@kripton indeed it is. We'll keep tracking this in #33 :)

closed 2022-10-25 by jostlowe