jostlowe/Pico-DMX #36: Sample Wiring

red4dj, 2022-11-27:
Could simple wiring diagrams be provided for those of us with a software background and not a hardware one? I would appreciate them as a guideline for your examples (input & 8 parallel output) and using them with MAX485 modules.

sonnny, 2022-12-31:
@red4dj 
see my youtube demo link below, source in the description

[youtube demo](https://www.youtube.com/watch?v=JEKsX28catU)

blockiyt, 2023-04-30:
Anything new?

jostlowe, 2023-05-19:
Unfortunately, no :(

My PhD thesis is due soon, so i will likely not be making any contributions in the near future

nsc-de, 2023-06-19:
For anyone still searching for answers on this topic: Here are two sample wirings for a common breakout module and the pi pico. The wiring should basically ve identical for other micro controllers. 
[image]
[image]

6by9, 2023-06-19:
@nsc-de Nice diagrams, but a couple of comments.

The Pico runs on 3.3V, but the standard MAX485 is a 5V device. You want to be using a [MAX3485](https://www.analog.com/en/products/max3485.html) when running on 3V3.
Other microcontrollers may be running on 5V, and there a MAX485 would be suitable.

Secondly your Receive DMX diagram shows GPIO1 being connected to DI. I think you meant to connect it to RO (Receiver Out) for receiving data.

nsc-de, 2023-06-19:
Most max485 chips work fine with 3.3V, but yeah, this may be a problem (and also for long range it is good practice to use the 5V instead of 3.3V). Instead of using another microcontroller you can use a transistor or stepup converter like this. This is quite a simple component to use, but it makes the diagram seem more complicated
[image]

You are right, tbh i just forgot to update the this part of the diagram while copy-pasting. I updated it in my first comment, its now correct.

jostlowe, 2023-06-19:
An ordinary MAX485 breakout should work fine when the VCC pin of the MAX is connected to the V_BUS pin on the Pico without a level converter. This is for two reasons: 

- If i remember correctly: The MAX485 reads anything over 2.2V as a logic high. The 3.3V gpio on the pico is sufficient to produce a logic high.
- The GPIO of the Pico is 5V-resistant. Any incoming data from the MAX to the Pico will have a 5V voltage level, but the GPIO is built to handle these voltage levels.

6by9, 2023-06-19:
> An ordinary MAX485 breakout should work fine when the VCC pin of the MAX is connected to the V_BUS pin on the Pico without a level converter. This is for two reasons:
>
>     * If i remember correctly: [...]

I must admit to not having checked that one, but you appear to be right that VIH is min 2V for DE, DI, and nRE according to the [datasheet](https://www.analog.com/media/en/technical-documentation/data-sheets/MAX1487-MAX491.pdf)

>     * The GPIO of the Pico is 5V-resistant. Any incoming data from the MAX to the Pico will have a 5V voltage level, but the GPIO is built to handle these voltage levels.

Officially it is not (I've just checked with colleagues here at Pi Towers).
[Datasheet](https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf) section 5.5.3
IOVDD: max 3.63V.
Input Voltage High @ IOVDD=3.3V: max IOVDD + 0.3, which will be max 3.93V.

Unofficially, it will stress the GPIOs rather than immediately latch up or fail, but it can't be recommended.

SutliffeProductions, 2023-06-19:
On the plus side, logic level shifters are cheap and easy to source.  Which is convenient. There are 3.3v rs485 modules out there, but I've not managed to get any of them to receive dmx yet... Probably my fault somewhere

6by9, 2023-06-19:
> On the plus side, logic level shifters are cheap and easy to source. Which is convenient. There are 3.3v rs485 modules out there, but I've not managed to get any of them to receive dmx yet... Probably my fault somewhere

Strange as there isn't much to get wrong as long as nRE and DE are pulled low as @nsc-de shows in his diagrams.
I'll have to give mine a go for receive (https://www.amazon.co.uk/DollaTek-module-SP3485-communication-Breakout/dp/B099DNYC3H).
Although with jostlowe's observation that the MAX485 has a VIH of 2V you don't need a level shifter for TX, and a simple potential divider would do for RX.

SutliffeProductions, 2023-06-19:
> Strange as there isn't much to get wrong as long as nRE and DE are pulled low as @nsc-de shows in his diagrams.
> I'll have to give mine a go for receive [...]

Ahh yeah, the modules I've been using are automatic switching so I shouldn't have to pull any pins high or low supposedly.  (From what I can tell, as there are no pins other than the gnd, vcc and tx and rx

6by9, 2023-06-19:
> Ahh yeah, the modules I've been using are automatic switching so I shouldn't have to pull any pins high or low supposedly. (From what I can tell, as there are no pins other than the gnd, vcc and tx and rx

Good luck with those.
From days of using Westermo MA-44 RS232 to RS485 transceivers, you generally have to tell them the baud rate you're using. They then have a monostable that holds the transmitter active for the configured delay after the last edge in the data. How you do it without any configuration is an interesting concept, and may be relying on the line bias.

Then again if you're only receiving that should be their default state.

SutliffeProductions, 2023-06-19:
> Good luck with those.
> From days of using Westermo MA-44 RS232 to RS485 transceivers, you generally have to tell them the baud rate you're using. They then have a monostable that holds the [...]

Yeah they've been a pain in the ass so far.  Will send dmx just fine, no issues. But they just refuse to receive it, and I don't really want to have to do some weird serial stuff, but I fear I might have to learn