jostlowe/Pico-DMX #38: parallel output not works as expected with multiple buffers

ScobbyDoo, 2022-12-27:
I'm trying to write parallel dmx with different buffers instead of one buffer.
`
// In example one buffer was used to send dmx data.
// Ex for 3 Universe 
uint8_t universe0 [512+1];
uint8_t universe1 [512+1];
uint8_t universe2 [512+1];

universe0[0]  = 255;
universe0[1]  = 255;
universe0[2]  = 255;

dmxOutputs[i].write (universe0, UNIVERSE_LENGTH +1);
dmxOutputs[i].write (universe1, UNIVERSE_LENGTH +1);
dmxOutputs[i].write (universe2, UNIVERSE_LENGTH +1);
`

jostlowe, 2022-12-27:
Hi! Can you please elaborate on how the parallell outputs behave unexpectedly?

Kamal-Sonani, 2022-12-27:
universe0[0] = 255; 
universe1[1] = 255;
universe2[2] = 255;

dmxOutputs[i].write (universe0, UNIVERSE_LENGTH +1);
dmxOutputs[i].write (universe1, UNIVERSE_LENGTH +1);
dmxOutputs[i].write (universe2, UNIVERSE_LENGTH +1);

i've 3 dmx instance with 1 channel size each on 1,2,3 
as i'm tacking 3 buffer , on the other side lights get blinking instead of constant ON
that's not hardware issue because with same buffer its glows constant without flickering on every gpio(universe)

ScobbyDoo, 2022-12-27:
universe1[1] = 255;
universe2[2] = 255;

dmxOutputs[i].write (universe0, UNIVERSE_LENGTH +1);
dmxOutputs[i].write (universe1, UNIVERSE_LENGTH +1);
dmxOutputs[i].write (universe2, UNIVERSE_LENGTH +1);

i've 3 dmx instance with 1 channel size each on 1,2,3
as i'm tacking 3 buffer , on the other side lights get blinking instead of constant ON
that's not hardware issue because with same buffer its glows constant without flickering on

jostlowe, 2022-12-28:
Ok, I see. Can you please include the full code of your Arduino sketch or Pico SDK project?

closed 2023-03-09 by jostlowe