jostlowe/Pico-DMX #40: how to define START_CHANNEL during runtime?

stephanschulz, 2023-02-06:
Thank you for making this library. 

I am starting the process of converting some of my Teensy code to work on a PICO. 
My project reads a dip switch to determine the DMX start channel.

In you example you define `volatile uint8_t buffer[DMXINPUT_BUFFER_SIZE(START_CHANNEL, NUM_CHANNELS)];` befor the code starts. 
Is there a way to either change the buffer or set the buffer during `void setup()` ?

Thank you for your advice.

jostlowe, 2023-03-09:
Hi, and sorry for the late reply!

Yes, you can re-initialize a DMX input, but as you said, you need to call `dmxInput.end()` before calling `dmxInput.begin(...)` again

michaelglass, 2024-02-14:
the other approach is reading the whole universe (all 512 channels) and dynamically deciding which channels to use. (from https://github.com/jostlowe/Pico-DMX/issues/15#issuecomment-1278197448). If your DMX device is sending a partial universe (which my fader does) then this is the only way of changing start channel anyways