jostlowe/Pico-DMX #44: write not sending channel 512

P4LIL, 2023-03-20:
setting num channels to 512 with 
dmxOutput.write(workingBuffer, NUM_CHANNELS); 
where NUM_CHANNELS is 512, doesnt seem to send channel 512, but 
dmxOutput.write(workingBuffer, NUM_CHANNELS+1); 
does.
small issue
thanks
Paul

jostlowe, 2023-03-20:
Hi!

Take a look at the examples at [the main page](https://github.com/jostlowe/Pico-DMX). You should add 1 to the number of channels you want to send to account for the start code of the DMX-frame. If you want to send a full-sized universe of 512 channels, the size of your working buffer should be 513, where the 0'th channel is your start code

closed 2023-03-20 by jostlowe

P4LIL, 2023-03-20:
thanks, my mistake, was using your example file highlight_all