jostlowe/Pico-DMX #45: Confirmed working with pico w?

SutliffeProductions, 2023-04-11:
Hi, sorry to be posting another issue, has anybody had this library working with the pico w?
I’ve ran it on many standard picos, and had nearly no issues but today I’ve been trying to run it on a pico w and have had no luck.  I’ve not got my oscilloscope with me at the moment, but I thought before I jump down a rabbit hole of debugging I’d ask if I was waisting my time.

joelschoene, 2023-04-12:
Please send your code lines. Maybe you're using a pin, that isn't available at Pico w, because it might be connected to wifi/bt modul.

SutliffeProductions, 2023-04-13:
> Please send your code lines. Maybe you're using a pin, that isn't available at Pico w, because it might be connected to wifi/bt modul.

Hi, sorry, turns out the new board config I was using in the IDE was defining pin numbers differently between the Pico and Pico w (Pico w profile was using numbers as GPIO numbers, whereas Pico was using it as pin numbers)
Sorry to waste your time