jostlowe/Pico-DMX #47: How to behave as a DMX USB Pro Widget ?

stephanschulz, 2023-05-25:
Is it possible use the Pico-DMX library to implement something like this USBProWidget which was written for Teensy?
https://github.com/ssilverman/TeensyDMX/tree/master/examples/USBProWidget

This would enable us to use the RPI-Pico as a drop in replacement for the ENTTEC USB Pro.

Thank you for any advice.

jostlowe, 2023-05-25:
Yes, I don't see why this shouldn't be possible to implement :)

The missing piece would be the USB-based communication between the Pico and the PC. I would love to see something like this built on top of PicoDMX

stephanschulz, 2023-05-25:
I am particularly interested in the USB receiving and DMX transmitting part of it. 
Thanks for confirming my gut feeling. 
I think this is the function handling the incoming USB data https://github.com/ssilverman/TeensyDMX/blob/master/examples/USBProWidget/USBProWidget.ino#LL331C6-L331C21

But there is a lot of code related to UART receiving that is a bit over my head: https://github.com/ssilverman/TeensyDMX/blob/master/src/UARTReceiveHandler.cpp

I will first have to learn how Teensy and Pico handle this respectively. 

Thanks again for your reply.

kripton, 2023-05-25:
Please be aware that the Teensy-Code you posted uses the "regular Teensy USB serial approach". I don't know which exactly this is (since I don't have much experience with Teensies and I didn't dig too deep into the code). However, it will look to the PC as a serial port with a "custom" or cloned VID/PID combination. As long as your DMX-generating software talks to a serial port with the USB Pro protocol, your approach will work.

However, the USB Pro uses an FTDI USB -> serial chip. There is some DMX generating software, that uses libftdi to talk to the chip directly, bypassing your OS's USB serial port code. Such a software will fail to see/talk to the Teensy since it simply does not behave like an FTDI chip.

The last link you posted is the Teensy-Code to receive DMX data using their UART peripheral. It's pretty low-level any uses the "UART frame error" detection to detect the end-of-frame condition. This is quite clever and the common approach if you want to use a "hardware UART" for DMX reception. This might also work on the Pico. However, Pico-DMX uses the PIOs to receive DMX and those behave completely different compared to a hardware UART. At the moment, DMX reception using Pico-DMX is not reliable in all cases. I am/was working on an improved solution (#21) but it's not yet done.

kripton, 2023-05-25:
If you are interested in PC via USB -> DMX, you don't need the UARTReceiveHandler.cpp code at all, that is for DMX -> PC via USB.
You would just need to implement the USB DMX Pro's protocol over your USB-Serial-code and maybe (depending on the software you use) to fake being an FTDI chip.

DaAwesomeP, 2024-08-25:
@stephanschulz just found this issue a bit late, but take a look at https://github.com/DaAwesomeP/dmxusb. You'll still need an external FTDI, but it implements the ENTTEC protocol.