jostlowe/Pico-DMX #49: This library in rust

JonasFocke01, 2023-06-07:
Hello there,

im tinkering around with my rpi-pico and want to set it up as a dmx receiver something.
Your library works great when in use with C/C++, but im also uinterested in the Rust lang.
I got my pico working in rust, and even write onto a WS2812b strip, but the dmx input part wont work.
I realy tried to work something out, that the garbage im receiving is interpreted as valid dmx, but i cant.
I tried to learn from the sourcecode of this lib, but im somewhat stuck..
The input im getting by setting the GPIO0 to uart with 250000 hz, eight data and two stop bits seems realy scuffed.
Are there any hints you could give me to succeed with this project? OC im willing to share any accomplishements i have made so far and will do in the future.
Thanks for the help!

JonasFocke01, 2023-06-08:
Thank you for the reply :heart: 
Its great to hear, that there is the possibility to get this working!
Im not yet that deep into the pico, so it blew my mind, that the assembly actually in some sort of a state machine, that can be used to configure the pio, and not just some auto generated compiler thingy! Im looking realy forward to tinker with this :smile: 
As im not thaat skilled of a embeded dev, that could take some time, but im on it!
And i will let you know how this went!

jostlowe, 2023-06-08:
Excellent! I'm looking forwards to hearing about your results :)

JonasFocke01, 2023-06-14:
So i got at least something working.
The sm seems to load the program (DMXInput), and starts reading. It in fact reads correctly, that a channel is set to something other than 0, but that is where the good news stop..
The program spits out random values (from 00 to ff).
I was able to confirm, that my max485 is configured correctly by trying it with an arduino.
I have hooked it up to gpio22 through a llc to get from 5V to the 3.3V.

Could you take a look at the code i currently have?
I suspect, that i have done something wrong with the clock frequency, but i dont realy know..
In my code is a USB driver configured, that writes the fifo result to usb to debug it.
```
#![no_std]
#![no_main]

use rp2040_hal::{clocks::init_clocks_and_plls, gpio::Pins, pac, sio::Sio};
use rp_pico::entry;
use hal::pio::PIOExt;
use panic_halt as _;
use rp_pico::hal;
use usb_device::{class_prelude::*, prelude::*};
use usbd_serial::SerialPort;

const XOSC_CRYSTAL_FREQ: u32 = 12_000_000;

#[entry]
fn main() -> ! {
    let mut pac = pac::Peripherals::take().unwrap();

    let mut watchdog = hal::Watchdog::new(pac.WATCHDOG);

    let sio = Sio::new(pac.SIO);
    let pins = Pins::new(
        pac.IO_BANK0,
        pac.PADS_BANK0,
        sio.gpio_bank0,
        &mut pac.RESETS,
    );
    let clocks = init_clocks_and_plls(
        XOSC_CRYSTAL_FREQ,
        pac.XOSC,
        pac.CLOCKS,
        pac.PLL_SYS,
        pac.PLL_USB,
        &mut pac.RESETS,
        &mut watchdog,
    )
    .ok()
    .unwrap();

    // Setup PIO
    pins.gpio22.into_pull_down_input();

    let program_with_defines = pio_proc::pio_file!("DmxInput.pio", select_program("DmxInput"),);

    let (mut pio, sm0, _, _, _) = pac.PIO0.split(&mut pac.RESETS);
    let installed = pio.install(&program_with_defines.program).unwrap();
    let (sm, mut rx, _) = rp2040_hal::pio::PIOBuilder::from_program(installed)
        .jmp_pin(22)
        .in_pin_base(22)
        .clock_divisor_fixed_point(48, 0)
        .build(sm0);
    sm.start();

    // Set up the USB driver
    let usb_bus = UsbBusAllocator::new(hal::usb::UsbBus::new(
        pac.USBCTRL_REGS,
        pac.USBCTRL_DPRAM,
        clocks.usb_clock,
        true,
        &mut pac.RESETS,
    ));

    let mut serial = SerialPort::new(&usb_bus);

    UsbDeviceBuilder::new(&usb_bus, UsbVidPid(0x16c0, 0x27dd))
        .manufacturer("Fake company")
        .product("Serial port")
        .serial_number("TEST")
        .device_class(2)
        .build();

    loop {
        if let Some(channel) = rx.read() {
            channel.to_le_bytes().iter().for_each(|channel| {
                if *channel != 0 {
                    serial.write(&channel.to_le_bytes()).unwrap();
                }
            });
        }
    }
}
```

jostlowe, 2023-06-26:
Hi! Have you had any progress on this?

My first immediate thoughts are these:
- Are you absolutely certain that your clock divider is correct? The clock for the PIO must be at _exactly_ 1MHz for the state machine to synchronize nicely with the incoming DMX signal
- The RX and TX buffers can be configured with a word size of 8, 16 or 32 bits. Are you certain that you have configured the RX buffer so that it reads 8 bit words? I believe the default is 32.
- For PicoDMX the state machine has to be reset after each frame is received. Is this done in your program?

JonasFocke01, 2023-06-27:
Hi!

So i made some progress, i think, but it feels like, that the path is a steep one and im far from finished :laughing:   
I double, triple, quadruple checked the configured clock speed and read a ton about this on the internet, and i think, this is correct now. The stuff it reads is still garbage, but consistent garbage :partying_face: , like 'one channel on 100 means it reads 30 indefinitely, but if i move the channel to 101, it suddenly reads 213 indefinitely' (those are random numbers and change each time i try it)  
Same occurs with two or three channels...  
I also tried to figure out how i could set a word-bit-size, but i was not able to. Im dividing the u32 that the read() function on the rx buffer gives me into four bytes anyway, so im a bit curious, what this would do?
For the last bulletpoint, i added an index to my program, that counts to 512 and restarts the sm, but that seems to make the problem worse again, as that adds some significant noise to what the pico is reading.

Can you make any sense of that? Does that sound familiar in some way to you?

jostlowe, 2023-07-04:
Hi!

Maybe this can be of help:
https://github.com/jostlowe/picodmx-rs

Note that this in its infancy and is very much a WIP.