jostlowe/Pico-DMX #50: Help with c++

sticckky1, 2023-06-08:
Hello! ,    Absolutely love your work! Thankyou for making this.     I have this working well with the arduinoIDE   but i need to get it working in VScode on the pico.  Do you have any suggestions for me at all? I'm very new to Cmake.  I am able to struggle through code with an example, but i am failing  at adding your library to my current project.   
Every time i incluide the DmcOutput.h or  .cpp files it tells me i'm missing dma.h or pio.h   

Thanks for any help you can provide.

jostlowe, 2023-06-08:
Unfortunately I'm absolutely rubbish with CMake myself!

Maybe @kripton who added the CMake support for this library can shed some light on your situation?

opsnlops, 2023-06-08:
Here's how I do it, if it helps: https://github.com/opsnlops/parrot-controller/blob/main/CMakeLists.txt#L17