jostlowe/Pico-DMX #51: Input and Output in the same program?

SutliffeProductions, 2023-06-08:
Hi, I really like this library, and have had no problems with it so far. but for my current project, i need to be able to both output and input dmx, (not at the same time, but without restarting the pico) can this be done by calling both functions, using 2 different pins with 2 different state machines? or do I have to end output before I can start input? also is there a pin predefined to pull high or low for switching the mode of rs485 modules, or do I need to code that in myself?

thanks,
Leon

jostlowe, 2023-06-10:
Heyo :)

Input and output at the same time is absolutely doable! If you are using separate pins for input and output, simply start two state machines, one for input and one for output. They should run fine in parallel 

Running input and output on the same pin is possible but somewhat more involved.

Pulling pins high or low to e.g. switch data direction on a transceiver is not included in Pico-DMX and has to be done manually. This is to keep Pico-DMX hardware agnostic :)

SutliffeProductions, 2023-06-10:
> Heyo :)
>
> Input and output at the same time is absolutely doable! If you are using separate pins for input and output, simply start two state machines, one for input and one for output. They should [...]

Great! Thanks for your help :)

closed 2023-06-10 by jostlowe