jostlowe/Pico-DMX #55: Adding RDM support

tomo7358, 2024-03-16:
Is there any chance that RDM support is planned?

jostlowe, 2024-03-18:
Hi! RDM support is not currently planned, but I'd be happy to provide some support for anyone who wishes to work on implementing RDM support!

closed 2024-03-18 by jostlowe

Redforce04, 2024-04-12:
> Hi! RDM support is not currently planned, but I'd be happy to provide some support for anyone who wishes to work on implementing RDM support!

Does the current implementation actively ignore RDM Packets? I was having an issue with random pulses that didn't make sense but I'm realizing it may have just been RDM packets. According to ETC:
> The problem isn’t often an issue at all with the RDM protocol. Instead, the issue typically rests in a fixture’s response to RDM packets. According to the DMX standard alone, all DMX packets must begin with a null start code (0x00). If packets start with something else, they should be ignored. RDM builds on top of the DMX standard and passes its data on an alternate start code (0xCC). The theory is then that any DMX device that does not speak RDM will simply ignore packets with any alternate start code.

The fix should be relatively easy, as it just has to ignore DMX packets that don't start with "0x00", and in theory it is protocol complaint.

RDM itself is cool, but might be a bit too heavy for the pico.

Redforce04, 2024-04-12:
@jostlowe (pinging you because I'm unsure of whether github notifies you about comments on closed issues)

jostlowe, 2024-04-12:
Hello! 

Pico-DMX is written to be completely agnostic with regards to e.g. RDM packets. When receiving or sending DMX frames, the zero'th byte of the sent or received frame indicates the start code of the frame. In your application you can choose to discard any frames where the zero'th byte in the frame is anything else than 0x00 indicating a "regular" DMX frame.

I guess you could also snoop out RDM frames by picking out frames with an 0xCC start code!

Tom-Wyz, 2024-04-13:
@Redforce04 if you end up figuring out how to get the rdm packets to work if you wouldn't mind sending me the code that would be much appreciated

Redforce04, 2024-04-15:
> @Redforce04 if you end up figuring out how to get the rdm packets to work if you wouldn't mind sending me the code that would be much appreciated 

I don't typically work in C/C++ but I do have some good C# / Dotnet experience. I can create an outline of changes & a rough layout / implementation, if you're willing to help adapt it to the library.

I'm pretty busy with a show right now but it seems like it would be a unique challenge that would help a lot of people.

Redforce04, 2024-04-15:
Looking at the protocol, it seems relatively simple. Implementing it with only 2 cores and the limited memory of the pi pico seems like it'd be a bigger problem. 

A majority of the protocol boils down to this:
[image]

The CC is the command type, PID is the Parameter (Message) Identifier, the PD is the Parameter (Message) Data

It also has a queue system so the device can respond / accept changes when it is less busy processing data. This would prove to be helpful with the low compute of the Pi Pico. However it is still a fight to keep up with only 2 cores, as it is easy to miss messages. Ideally one core would process and queue up dmx messages / rdm messages, while the other core would process the actual data / respond. 

One issue with this approach is that if the processing core gets backed up, it may buffer and process old frames, rather than responding to new frames. Maybe a TTL system would reduce this? (if > 4 frames are present, delete all frames except the last message). Every other frame could be dedicated to RDM (if present). 

Regardless, this would require a much more advance implementation. I think adding a static setting to prevent RDM frames from being passed through would be best. Maybe pair this with a separate read function for RDM frames. This would prevent RDM 'Blips' in the DMX Data for unknowing users in future implementations, without breaking current use cases.

Redforce04, 2024-04-15:
There are plenty of PID's for different message types, and they all have different PD data depending on the information being transferred.

Redforce04, 2024-04-15:
[image]
Here is the RDM Packet. It happens on a typical interval of ~33 frames (assuming my counting is right).

What my problem was actually isnt rdm. Rdm is a part of the problem but it really stems from a phantom input of some sort:
[image]

The flickering I am experience is actually not due to RDM. It seems to be beacause the pico reads too many channels too fast? Whenever I listen to 20 channels and ignore the 15, it doesnt "flicker". when I use just 5 channels, it seems to encounter the problem. Whatever the case I think adding a setting for RDM transparency would be nice. I'll see what I can whip up (but am quite busy with a show right now).