jostlowe/Pico-DMX #57: cant change the start channel

michaelkpierce, 2024-06-21:
Hi Jostein,
I've been using your library on a small project that I am working on. I works great when the starting channel is 1!
However, I don't seem to be able to move the Start_channel.  For instance I would like the starting channel to be 20. When I change the starting channel like so
#define START_CHANNEL 20
#define NUM_CHANNELS 4
Its still working with starting channel as 1.

Is this not supposed to be the way that it works.

I suppose I can try to work around by inputing all 512 channels and then only looking for the channels that I want in the buffer.

Thanks
Michael Pierce

SutliffeProductions, 2024-06-21:
I've not got my test rig in front of me to check this out right now, but can you post your code? Do you have this problem with the example programs?

michaelkpierce, 2024-06-21:
Leon,
Thanks for getting back to me.
I am using your sample code to test right now.

I have simply changed the start channel to 10 in this example but the unit
still responds to start channel 1.  Am i misunderstanding how it works?

See code below;
/*
 * Copyright (c) 2021 Jostein Løwer
 *
 * SPDX-License-Identifier: BSD-3-Clause
 *
 * Description:
 * Starts a DMX Input on GPIO pin 0 and read channel 1-3 repeatedly
 */

#include <Arduino.h>
#include "DmxInput.h"
DmxInput dmxInput;

#define START_CHANNEL 10
#define NUM_CHANNELS 3

volatile uint8_t buffer[DMXINPUT_BUFFER_SIZE(START_CHANNEL, NUM_CHANNELS)];

void setup()
{
    // Setup our DMX Input to read on GPIO 0, from channel 1 to 3
    dmxInput.begin(0, START_CHANNEL, NUM_CHANNELS);

    // Setup the onboard LED so that we can blink when we receives packets
    pinMode(LED_BUILTIN, OUTPUT);
}

void loop()
{
    // Wait for next DMX packet
    dmxInput.read(buffer);

    // Print the DMX channels
    Serial.print("Received packet: ");
    for (uint i = 0; i < sizeof(buffer); i++)
    {
        Serial.print(buffer[i]);
        Serial.print(", ");
    }
    Serial.println("");

    // Blink the LED to indicate that a packet was received
    digitalWrite(LED_BUILTIN, HIGH);
    delay(10);
    digitalWrite(LED_BUILTIN, LOW);
}

Thanks again
Regards
Michael

DaAwesomeP, 2024-09-08:
I'm having the same issue--no matter what it it seems to stick to channel 1.

DaAwesomeP, 2024-09-08:
Taking a look at [DmxInput.cpp](https://github.com/jostlowe/Pico-DMX/blob/c8ecde5a3267867a2cdcae0995fb8bd25f7280df/src/DmxInput.cpp), it doesn't look like `start_channel` or `_start_channel` are ever sent to the PIO or used beyond setting up the buffer.

michaelkpierce, 2024-09-08:
The workaround is to read the whole universe into memory (plenty of memory
in the PICO) and read the data you want.
MP

DaAwesomeP, 2024-09-08:
I see. I'm using FreeRTOS and trying to keep memory small per task.

This commented out line looks like the culprit that maybe @jostlowe left unfinished:

https://github.com/jostlowe/Pico-DMX/blob/c8ecde5a3267867a2cdcae0995fb8bd25f7280df/src/DmxInput.cpp#L93